pruwait/Nice_BusT4 #3: Buy me a beer / coffee does not accept foreign cards

xdanik, 2023-03-13:
Hello @pruwait ,

Many thanks for making this project!
You saved me a lot of time by figuring this protocol / sniffing the commands.

I wanted to thank you and buy you a beer/coffee.
Unfortunately yoomoney.ru seems to be only for domestic transfers inside Russia as it does not accept any of my European credit cards (including Revolut) - it says that foreign cards are not supported (I had to Google translate the error message...).

I would highly recommend you to create account on some platform with support for international payments.

pruwait, 2023-03-16:
use [paypal](https://www.paypal.com/paypalme/pruwait)

xdanik, 2023-03-16:
Tryed - "Sorry, this recipient can't accept payments right now."

[image]

pruwait, 2023-03-23:
Paypal not working for russians(
You can pay for my webhosting if you want. Foreign cards is accept but minimal payment is 1000RUR (about $13 today).
https://cp.vdsina.ru/operation/select/3b63f9f786d0c72b55a53873b95f32a72a65f7c