pruwait/Nice_BusT4 #6: Nice SPIN

podly, 2023-06-25:
Hi!

First of all thank you for great job you did!

I made a copy of this device and connected it to my garage door Nice SPIN. I can open and close gate, I can also send command for partial open and this is working fine. However there is no feedback on gate position other then fully close or fully open. If I send Partial Open 1 command, gates open partially, but ESP reports state closed and position 0. Setting position is also not working - only partial open commands works.

Second what I noticed is lack of stop button in Home Assistant entity - it shows only up and down arrow:

[image]

Info from ESP:
<code>
[22:01:24][C][bus_t4.cover:502]:   Bus T4 Cover
[22:01:24][C][bus_t4.cover:509]:   Type: Sectional door
[22:01:24][C][bus_t4.cover:524]:   Maximum encoder or timer position: 0
[22:01:24][C][bus_t4.cover:525]:   Opened gate position: 2048
[22:01:24][C][bus_t4.cover:526]:   Closed gate position: 0
[22:01:24][C][bus_t4.cover:529]:   Manufacturer: NICE 
[22:01:24][C][bus_t4.cover:532]:   Drive unit: SPINBUSR10 
[22:01:24][C][bus_t4.cover:535]:   Drive hardware: 598BR00 
[22:01:24][C][bus_t4.cover:538]:   Drive firmware: BZ01 
[22:01:24][C][bus_t4.cover:541]:   Drive description: NewSpin 
[22:01:24][C][bus_t4.cover:544]:   Gateway address: 0x0066
[22:01:24][C][bus_t4.cover:545]:   Drive address: 0x0003
[22:01:24][C][bus_t4.cover:546]:   Receiver address: 0x0000
[22:01:24][C][bus_t4.cover:549]:   Receiver:  
[22:01:24][C][bus_t4.cover:552]:   Receiver hardware:  
[22:01:24][C][bus_t4.cover:555]:   Receiver firmware:  
[22:01:24][C][bus_t4.cover:558]:   Receiver description:  
[22:01:24][C][bus_t4.cover:560]:   Auto close - L1: No 
[22:01:24][C][bus_t4.cover:561]:   Close after photo - L2: No 
[22:01:24][C][bus_t4.cover:562]:   Always close - L3: No 
[22:01:24][C][captive_portal:088]: Captive Portal:
[22:01:24][C][web_server:151]: Web Server:
[22:01:24][C][web_server:152]:   Address: nice-bust4-uart.local:80
[22:01:24][C][mdns:108]: mDNS:
[22:01:24][C][mdns:109]:   Hostname: nice-bust4-uart

</code>

I also see also in ESP logs that sometimes checksum errors are reported:

<code>
[20:26:35][D][button:010]: 'Step by step' Pressed.
[20:26:35][I][bus_t4.cover:660]: Sent: 55.0C.00.03.00.66.01.05.61.01.82.01.64.E6.0C (15) 
[20:26:35][I][bus_t4.cover:157]: Package received: 55.0E.FF.FF.00.03.01.07.05.04.02.84.19.00.00.9B.0E (17) 
[20:26:35][D][bus_t4.cover:366]: RSP package received
[20:26:35][I][bus_t4.cover:369]: HEX data 19.00 
[20:26:35][I][bus_t4.cover:372]: Drive controller package
[20:26:35][I][bus_t4.cover:376]: RUN submenu
[20:26:35][I][bus_t4.cover:391]: Command: CLOSE
[20:26:35][I][bus_t4.cover:437]: Operation: 84
[20:26:35][D][cover:170]: 'Nice SPIN' - Publishing:
[20:26:35][D][cover:173]:   Position: 0%
[20:26:35][D][cover:186]:   Current Operation: CLOSING
[20:26:37][W][bus_t4.cover:126]: Received invalid message checksum 1 01!=58
[20:26:46][I][bus_t4.cover:157]: Package received: 55.0E.FF.FF.00.03.01.07.05.04.02.05.19.00.00.1A.0E (17) 

</code>

and sometimes operations ends with timeout:

<code>
[20:38:48][D][button:010]: 'Step by step' Pressed.
[20:38:48][I][bus_t4.cover:660]: Sent: 55.0C.00.03.00.66.01.05.61.01.82.01.64.E6.0C (15) 
[20:38:49][I][bus_t4.cover:157]: Package received: 55.0E.FF.FF.00.03.01.07.05.04.02.86.19.00.00.99.0E (17) 
[20:38:49][D][bus_t4.cover:366]: RSP package received
[20:38:49][I][bus_t4.cover:369]: HEX data 19.00 
[20:38:49][I][bus_t4.cover:372]: Drive controller package
[20:38:49][I][bus_t4.cover:376]: RUN submenu
[20:38:49][I][bus_t4.cover:402]: Operation completed with timeout
[20:38:49][I][bus_t4.cover:437]: Operation: 86
[20:38:56][D][button:010]: 'Step by step' Pressed.

</code>

any chance that you can help me with this?

br,

Michal

feeleep75, 2023-11-25:
@podly - maybe you can share an EasyEDA project or full BOM at least?

pruwait, 2024-01-22:
Not all gate drives send their position during the maneuver. The spin is not sent in your logs.
When turned on for the first time, initialization occurs and the drive receives command 11. 
It responds with its position.
Can you send command 11 and show the log?

closed 2024-02-10 by pruwait