pruwait/Nice_BusT4 #11: OVIEW firmware

gashtaan, 2024-06-16:
Hi, sorry that I'm contacting you here, but I didn't find other way how to reach you. I've noticed you've forked my repository related to Nice BiDi-WiFi and you are looking to its firmware too. Any chance you got hands on OVIEW firmware too? Unfortunately I don't have this device in my hands, so I can't help myself, but I believe it could help me to achieve a better understanding of various aspects of T4 protocol.

Egyras, 2024-06-16:
Hi, problem that I dont have Oview for any testing.... :(

gashtaan, 2024-06-18:
@Egyras My comment was meant for @pruwait as he sniffed the OVIEW communication, so he obviously has the device... or at least he did at that time.

But never mind I've already found out what I wanted from reversing the Nice Pro app and RBA3R10 firmware.

pruwait, 2024-07-28:
I haven't oview firmware, only device.

closed 2024-07-29 by gashtaan